firesim/FireMarshal #43: Optionally re-build from changed dependencies

NathanTP, 2019-07-10:
In an abundance of caution, marshal currently does not inherit changes from dependencies if you don't clean first. This is to avoid wiping out ephemeral data on your rootfs accidentally. It's not a big deal to just "marshal clean work.json && marshal build work.json" if you want the new dependencies, but a "marshal build --force" might be more clear. This would re-inherit dependencies all the way down the chain if needed, but will remove any ephemeral data you had laying around in your rootfs.

NathanTP, 2019-10-07:
thinking about this some more, I think there should be a warning or "type yes to continue" type of thing if a parent changed.

NathanTP, 2020-12-11:
After more discussion there seem to be a few options on how to address this:

1. Add an interactive warning if marshal is going to overwrite your workload. e.g. "Parent workloads have changed. Overwrite your workload image with the new parent (all local changes will be lost)? Y/N"
    1. May lead to annoying behavior when you try to 'fire and forget', it makes 'marshal build' potentially interactive and could break scripts.
2. Change the default behavior to just always overwrite (how common is it for people to have important manual changes in their workload?). I'm not sure exactly how Docker handles this.
    1. Could cause some annoying surprises from time to time (if you were in the middle of manually tweaking the workload). Honestly I don't think this is very common anymore, everything should be in your workload config and it's pretty fast to rebuild these days.
3. Keep the current default behavior, but add a "--force" flag that will overwrite the child on parent changes. There would also be a new warning that parent changes were not applied and to use '--force' if you're willing to lose local changes. This seems similar to other tools like git.
    1. Is probably the safest, but I suspect most people actually want the 'force' behavior as the default and everyone will just do 'marshal build -f' all the time anyway.

Opinions?

a0u, 2020-12-11:
> This is to avoid wiping out ephemeral data on your rootfs accidentally.

Is this a concern about overwriting a parent's rootfs image that has been modified by the user after build time (e.g., customized through a running system)?  Personally, I just assume that any file generated by marshal can potentially be clobbered anytime I run `marshal build`.  It's usually more confusing to me when something _doesn't_ change as expected through the dependency tree.