firesim/FireMarshal #92: [dev] Bad relative path for symlinks

abejgonzalez, 2019-10-31:
When running `./marshal build example-workloads/sha3-linux.json` in `dev` I get the following error: 

```
ValueError: host-init script /home/centos/general/firemarshal/example-workloads/sha3/marshal-configs/sha3/build.sh not found.
```

`sha3-linux.json` is a symlink to a json in `marshal-configs` within the `workdir` which seems to indicate that the workdir is being specified from the `.json` location not where the symlink is.

abejgonzalez, 2019-10-31:
The simple solution to this is to make the `workdir == ..` but I don't know if that is the preferred solution.

NathanTP, 2019-11-07:
This is the expected behavior. Although I may reconsider the defaults now that marshal can build anywhere. It may make more sense to have the workdir default to '.'

closed 2019-11-07 by NathanTP