firesim/FireMarshal #109: qemu-system-riscv64: cannot set up guest memory 'riscv_virt_board.ram': Cannot allocate memory

smuchand, 2019-12-01:
Hello, 
 I am beginner here and was following the documentation to boot Linux kernal using the buildroot-based linux distribution provided in the repo. I could build the workload cleanly but I am facing the below issue when I try to run it in qemu. 
Command used to run: ./marshal -v launch workloads/br-base.json
************************
Running: qemu-system-riscv64 -nographic -bios none -smp 4 -machine virt -m 16384 -kernel /<path>/chipyard/sims/firesim/sw/firesim-software/images/br-base-bin -object rng-random,filename=/dev/urandom,id=rng0 -device virtio-rng-device,rng=rng0 -device virtio-net-device,netdev=usernet -netdev user,id=usernet,hostfwd=tcp::48179-:22 -device virtio-blk-device,drive=hd0 -drive file=/<path>/chipyard/sims/firesim/sw/firesim-software/images/br-base.img,format=raw,id=hd0 
qemu-system-riscv64: cannot set up guest memory 'riscv_virt_board.ram': Cannot allocate memory
************************
Can anyone help me with this? Thanks!

NathanTP, 2019-12-02:
It may be that your host machine doesn't have enough memory to handle the default target memory of 16GB (this is set to match FireSim's default btw).

If you are on the dev branch of FireMarshal, you can change this using the [`mem` option](https://firemarshal.readthedocs.io/en/latest/workloadConfig.html#mem) in your workload configuration json.

If you are on master or an older branch that lacks the mem option, you can manually change the default in wlutil/launch.py (it's the `launch_mem` variable near the top)

smuchand, 2019-12-05:
Ohh thanks, yes it was the host memory issue. I made the changes and it worked. Thanks :)

closed 2019-12-05 by smuchand