firesim/FireMarshal #114: subprocess.CalledProcessError: Command 'make -j2' returned non-zero exit status 2

HaFred, 2019-12-16:
When I tried to run marshal to build sha3 Bare-Metal test with:
./marshal build example-workloads/sha3/marshal-configs/sha3-bare-rocc.json

Errors occurred as below:
`
Traceback (most recent call last):
  File "./marshal", line 170, in <module>
    main()
  File "./marshal", line 67, in main
    wlutil.initialize()
  File "/home/frederick/workspace/chipyard/sims/firesim/sw/firesim-software/wlutil/wlutil.py", line 85, in initialize
    run(['make', jlevel], cwd=busybox_dir)
  File "/home/frederick/workspace/chipyard/sims/firesim/sw/firesim-software/wlutil/wlutil.py", line 180, in run
    raise sp.CalledProcessError(p.returncode, prettyCmd)
subprocess.CalledProcessError: Command 'make -j2' returned non-zero exit status 2.`

HaFred, 2019-12-16:
I tried the solution in https://github.com/firesim/FireMarshal/issues/30#issue-452667308, but it didn't work. Would you mind taking a look? @NathanTP 
Thank you very much.

HaFred, 2019-12-18:
Could it be a bug here in this python file?   `../wlutil/wlutil.py`

NathanTP, 2019-12-18:
Which branch is this on? That trace looks like it might be from an old branch (or maybe master). Does this issue show up on the dev branch?

HaFred, 2019-12-19:
Yes, it was from master branch. So I checked out the dev branch by `git checkout dev`, and reran the marshal.
```
frederick@ubuntu:~/workspace/chipyard/software/firemarshal$ git checkout dev
warning: unable to rmdir 'workloads/sha3': Directory not empty
M	boards/firechip/drivers/iceblk-driver
M	boards/firechip/drivers/icenet-driver
M	riscv-linux
M	riscv-pk
M	wlutil/br/buildroot
Switched to branch 'dev'
Your branch is up to date with 'origin/dev'.
frederick@ubuntu:~/workspace/chipyard/software/firemarshal$ ./marshal build workloads/sha3-bare-rocc.json 
sudo: a password is required
To check on progress, either call marshal with '-v' or see the live output at: 
/home/frederick/workspace/chipyard/sims/firesim/sw/firesim-software/logs/sha3-bare-rocc-build-2019-12-19--08-39-11-6MGU1X0KGLLCXAWN.log
Traceback (most recent call last):
  File "./marshal", line 201, in <module>
    main()
  File "./marshal", line 116, in main
    targetCfg = cfgs[cfgName]
  File "/home/frederick/workspace/chipyard/sims/firesim/sw/firesim-software/wlutil/config.py", line 432, in __getitem__
    return self.cfgs[key]
KeyError: 'sha3-bare-rocc.json'
```
Obviously, the `.json` file is gone so I re-checkout the `/workloads` from the master branch like below:
```
frederick@ubuntu:~/workspace/chipyard/software/firemarshal$ git checkout master -- workloads/
frederick@ubuntu:~/workspace/chipyard/software/firemarshal$ ./marshal build workloads/sha3-bare-rocc.json 
sudo: a password is required
To check on progress, either call marshal with '-v' or see the live output at: 
/home/frederick/workspace/chipyard/sims/firesim/sw/firesim-software/logs/sha3-bare-rocc-build-2019-12-19--08-40-25-ZR6E6AVE810PJJT1.log
.  /home/frederick/workspace/chipyard/sims/firesim/sw/firesim-software/workloads/sha3/build.sh 
Applying host-init: /home/frederick/workspace/chipyard/sims/firesim/sw/firesim-software/workloads/sha3/build.sh 
TaskError - taskid:/home/frederick/workspace/chipyard/sims/firesim/sw/firesim-software/workloads/sha3/build.sh 
PythonAction Error
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/doit/action.py", line 437, in execute
    returned_value = self.py_callable(*self.args, **kwargs)
  File "/home/frederick/workspace/chipyard/sims/firesim/sw/firesim-software/wlutil/build.py", line 46, in handleHostInit
    run([config['host-init'].path] + config['host-init'].args, cwd=config['workdir'])
  File "/home/frederick/workspace/chipyard/sims/firesim/sw/firesim-software/wlutil/wlutil.py", line 417, in run
    raise sp.CalledProcessError(p.returncode, prettyCmd)
subprocess.CalledProcessError: Command '/home/frederick/workspace/chipyard/sims/firesim/sw/firesim-software/workloads/sha3/build.sh' returned non-zero exit status 2.

Failed to build workload sha3-bare-rocc.json
Log available at: /home/frederick/workspace/chipyard/sims/firesim/sw/firesim-software/logs/sha3-bare-rocc-build-2019-12-19--08-40-25-ZR6E6AVE810PJJT1.log
```
Any ideas on this? Thanks again.

NathanTP, 2019-12-19:
The sha3 workload has moved to https://github.com/ucb-bar/sha3/tree/master/software.

The README there is mostly accurate. The only difference is that it asks you to run marshal as "./marshal", but of course you'll need to replace that with whatever the path to your marshal command is (or put marshal on your $PATH).

Also, it may help to attach the full log of the run (e.g. /home/frederick/workspace/chipyard/sims/firesim/sw/firesim-software/logs/sha3-bare-rocc-build-2019-12-19--08-40-25-ZR6E6AVE810PJJT1.log) since it contains more information. The output shown here just says that the host-init script (build.sh) failed but not why. You can try running the script manually to see more clear output for what is wrong.

HaFred, 2019-12-20:
You are right, Nathan. Problem solved with `build.sh` in sha3 workload.
Thanks.

closed 2019-12-20 by HaFred