firesim/FireMarshal #154: RISCV tools missing error

yunusdawji, 2020-05-24:
I have riscv tools installed and is added to path. But when i run marshal i get an error complaining riscv tools missing.

$./marshal
No riscv toolchain detected. Please install riscv-tools.

NathanTP, 2020-05-25:
The error check is on line 12 of the marshal command. Specifically, it's looking for 'riscv64-unknown-linux-gnu-gcc' which is the compiler it uses for almost everything. I'd try running "which riscv64-unknown-linux-gnu-gcc" in the same shell you're using to call marshal, if it doesn't find it, then there is something wrong with you're riscv-tools setup.

yunusdawji, 2020-05-25:
Thanks for the reply. I found that risc toolchain installed through homebrew was missing the linux version. I changed the path to manually built version and now it works fine. It would be better if it tells you exactly which toolchain is missing.

closed 2020-05-25 by yunusdawji