firesim/FireMarshal #198: Can't generators images

yangshuaixian, 2021-07-14:
Type the command ./marshal -d build ./boards/firechip/base-workloads/br-base.json
Return ERROR: Failed to build workload br-base.json
[image]
Open ./boards/default/distros/br/buildroot/output/build/ncurses-6.1/config.log
[image]
After executing the command with the change soft link, it is still the same.

NathanTP, 2021-07-14:
It seems like buildroot is having some issues, it's not the most robust system in the world. Marshal should clean it before attempting to build but some things may not be cleaned correctly (by buildroot). You can try running "git clean -fdx" in boards/firechip/distros/br/buildroot/ or starting from a completely clean fresh clone of FireMarshal. If you customized br-base.json in any way, make sure to undo that.

Another thing that might not be right here is your toolchain install. You could try re-installing that (I only test with the toolchain [installed by chipyard](https://chipyard.readthedocs.io/en/latest/Chipyard-Basics/Initial-Repo-Setup.html#building-a-toolchain)).

As a side note, you can run "marshal build br-base.json", no need to pass the whole path.

yangshuaixian, 2021-07-16:
Thanks for your help. I can generate IMG files now. One thing I have to tell you is that init-submodules.sh has something wrong. They can’t git submodule update - - init boards/firechip/drivers/* in my computer. I git clone drivers manually. And it can work now.

NathanTP, 2021-07-16:
How did it fail to clone the drivers? What was the error?

yangshuaixian, 2021-07-17:
When I init the submodes, it didn’t show something error(Or not in the bottom). And run ./marshal build br-base.json, one mistake is that make clean no target(the detail I forgot), Then I was looking for it. The driver folder was found to be empty. I forgot to keep the error messages. Sorry.

NathanTP, 2021-08-24:
I'm not sure what the issue could be without more information. init-submodules.sh just calls "git submodule update --init" and I would expect it to throw an error if something went wrong. Whatever went wrong was likely either a transient network or github error or something wrong with git (git can get itself in a pretty bad state sometimes if you move files or delete the wrong thing). 

The original issue seems resolved so I'll close the issue.

If this happens again feel free to file a new bug. Try to record any errors and also manually run "git submodule update --init" to see if it prints more errors or fixes the issue.

closed 2021-08-24 by NathanTP