firesim/FireMarshal #202: firemarshal build linux image for the FPGA prototype failed

zhushaofeng1992, 2021-08-16:
when execute
~/chipyard/software/firemarshal$ ./marshal -v -d build br-base.json
i got error

```
DEBUG: >>> gdb 10.1 Configuring
DEBUG: (cd /home/zsf/chipyard/software/firemarshal/boards/default/distros/br/buildroot/output/build/gdb-10.1/gdb/gdbserver && rm -rf config.cache && ...
DEBUG: bin/bash: line 0: cd: /home/zsf/chipyard/software/firemarshal/boards/default/distros/br/buildroot/output/build/gdb-10.1/gdb/gdbserver: No such file or directory
DEBUG: package/pkg-generic.mk:249: recipe for target '/home/zsf/chipyard/software/firemarshal/boards/default/distros/br/buildroot/output/build/gdb-10.1/.stamp_configured' failed

```
i thought because the GDB 10.1 has move the gdbserver at the top-level, so i edit the
~/chipayrd/software/firemarshal/boards/prototype/distros/br/buildroot/package/gdb/gdb.mk
change "GDB_SUBDIR = gdb/gdbserver to GDB_SUBDIR = gdbserver"

```

GDB_VERSION = $(call qstrip,$(BR2_GDB_VERSION))
GDB_SITE = $(BR2_GNU_MIRROR)/gdb
GDB_SOURCE = gdb-$(GDB_VERSION).tar.xz

# recent gdb versions (>= 10) have gdbserver moved at the top-level,
# which requires a different build logic.
ifeq ($(BR2_GDB_VERSION_10),y)
GDB_GDBSERVER_TOPLEVEL = y
endif

ifeq ($(BR2_arc),y)
GDB_SITE = $(call github,foss-for-synopsys-dwc-arc-processors,binutils-gdb,$(GDB_VERSION))
GDB_SOURCE = gdb-$(GDB_VERSION).tar.gz
GDB_FROM_GIT = y
# recent gdb versions (>= 10) have gdbserver moved at the top-level,
# which requires a different build logic.
GDB_GDBSERVER_TOPLEVEL = y
endif

ifeq ($(BR2_csky),y)
GDB_SITE = $(call github,c-sky,binutils-gdb,$(GDB_VERSION))
GDB_SOURCE = gdb-$(GDB_VERSION).tar.gz
GDB_FROM_GIT = y
endif

GDB_LICENSE = GPL-2.0+, LGPL-2.0+, GPL-3.0+, LGPL-3.0+
GDB_LICENSE_FILES = COPYING COPYING.LIB COPYING3 COPYING3.LIB
GDB_CPE_ID_VENDOR = gnu

# On gdb < 10, if you want to build only gdbserver, you need to
# configure only gdb/gdbserver.
ifeq ($(BR2_PACKAGE_GDB_DEBUGGER)$(GDB_GDBSERVER_TOPLEVEL),)
GDB_SUBDIR = gdb/gdbserver # change to gdbserver
```
after execute
~/chipyard/software/firemarshal$ ./marshal -v -d build br-base.json
i still got

```
EBUG: configure: WARNING: unrecognized options: --disable-gtk-doc, --disable-gtk-doc-html, --disable-doc, --disable-docs, --disable-documentation, --with-xmlto, --with-fop, --disable-dependency-tracking, --enable-ipv6, --disable-nls, --disable-static, --enable-shared, --without-uiout, --disable-gdbtk, --without-x, --disable-sim, --disable-binutils, --disable-install-libbfd, --disable-ld, --disable-gas, --disable-gprof, --without-included-gettext, --enable-static, --without-mpfr, --disable-gdb, --without-curses, --enable-gdbserver, --disable-tui, --without-python, --without-expat, --without-lzma, --without-zlib
DEBUG: >>> gdb 10.1 Building
DEBUG: PATH="/home/zsf/chipyard/software/firemarshal/boards/default/distros/br/buildroot/output/host/bin:/home/zsf/chipyard/software/firemarshal/boards/default/distros/br/buildroot/output/host/sbin:/home/zsf/chipyard/riscv-tools-install/bin:/home/zsf/chipyard/software/firemarshal:/home/zsf/chipyard/bin:/home/zsf/chipyard/software/firemarshal:/home/zsf/chipyard/bin:/home/zsf/chipyard/software/firemarshal:/home/zsf/chipyard/bin:/home/zsf/bin:/home/zsf/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/usr/local/cmake/bin" gl_cv_func_gettimeofday_clobber=no gl_cv_func_working_strerror=yes gl_cv_func_strerror_0_works=yes gdb_cv_prfpregset_t_broken=no /usr/bin/make -j2 MAKEINFO=true -C /home/zsf/chipyard/software/firemarshal/boards/default/distros/br/buildroot/output/build/gdb-10.1/gdbserver
DEBUG: make[2]: *** No rule to make target '../gnulib/import/libgnu.a', needed by 'gdbserver'.  Stop.

```
How to fix this?

NathanTP, 2021-08-17:
I'm not sure exactly what is happening since it's a build issue with buildroot itself. That being said, our default config should work but Buildroot is a bit touchy and prone to breakage (sometimes intermittent due to network problems) . I recommend running "make clean" in boards/firechip/distros/br/buildroot, or even "git clean -fdx" in that same directory to make sure it hasn't messed itself up somehow. Also, make sure all submodules are updated recursively. Be sure to revert any changes you made to the buildroot config before trying again.

NathanTP, 2022-07-13:
Closing due to inactivity. If the problem re-surfaces, feel free to open a new issue.

closed 2022-07-13 by NathanTP