firesim/FireMarshal #204: Chipyard FireMarshal br-base workload on CentOS7 failing

michael-etzkorn, 2021-08-24:
All dependencies installed. Following chipyard's recommendations for building the workload. 
Ran
`./init-submodules.sh`
Ran python and centos installations.
Added `board-dir : 'boards/prototype'` to my `marshal-config.yaml`
Ran `./marshal -v -d build br-base.json # here the -d indicates --nodisk or initramfs` which failed with this output: 

```
DEBUG: You seem to have the current working directory in your
DEBUG: LD_LIBRARY_PATH environment variable. This doesn't work.
DEBUG: support/dependencies/dependencies.mk:27: recipe for target 'dependencies' failed
DEBUG: make[1]: *** [dependencies] Error 1
DEBUG: Makefile:84: recipe for target '_all' failed
DEBUG: make: *** [_all] Error 2
TaskError - taskid:/mnt/Vivado_part/chipyard/chipyard/software/firemarshal/images/br.bc88.img
PythonAction Error
Traceback (most recent call last):
  File "/home/_eda/.local/lib/python3.6/site-packages/doit/action.py", line 437, in execute
    returned_value = self.py_callable(*self.args, **kwargs)
  File "/mnt/Vivado_part/chipyard/chipyard/software/firemarshal/boards/prototype/distros/br/br.py", line 190, in buildBaseImage
    wlutil.run(['make'], cwd=br_dir / "buildroot", env=env)
  File "/mnt/Vivado_part/chipyard/chipyard/software/firemarshal/wlutil/wlutil.py", line 498, in run
    raise sp.CalledProcessError(p.returncode, prettyCmd)
subprocess.CalledProcessError: Command 'make' returned non-zero exit status 2.
```

I've sourced env.sh from Chipyard which sources env-riscv-tools.sh. This seems to be what sets LD_LIBRARY_PATH. Not sure why this wouldn't work. 
```
export CHIPYARD_TOOLCHAIN_SOURCED=1
export RISCV=/mnt/Vivado_part/chipyard/chipyard/riscv-tools-install
export PATH=${RISCV}/bin:${PATH}
export LD_LIBRARY_PATH=${RISCV}/lib${LD_LIBRARY_PATH:+":${LD_LIBRARY_PATH}"}                                                                             
```

NathanTP, 2021-08-24:
This appears to be a Buildroot limitation (buildroot's build system is a bit janky). You can print your LD_LIBRARY_PATH to make sure '.' isn't in it (some libraries do this, I remember a CAD tool once did it to me). We run that make command in the buildroot directory (boards/firechip/distros/br/buildroot) so make sure that didn't somehow end up on your LD_LIBRARY_PATH (bizzare but stuff happens).

If you want to be extra paranoid, you can add "print(env['LD_LIBRARY_PATH'])" in boards/firechip/distros/br/br.py:182 though that really should match what your shell has.

If that doesn't work, you can try running "git -fdx" in boards/firechip/distros/br/buildroot/ and try building again. Sometimes buildroot gets itself all confused and breaks.

michael-etzkorn, 2021-08-24:
Ah, it seems the env-riscv-tools is adding the current directory to the path with: `${LD_LIBRARY_PATH:+":${LD_LIBRARY_PATH}"} ` since it adds `::` between the riscv-tools-install library path and the rest of the path variables.

closed 2021-08-24 by michael-etzkorn