firesim/FireMarshal #215: How to get the FPGA prototype in Firemarshal？

L-Chocoland, 2021-11-08:
I'm trying to build FPGA prototype in my own VMware linux.
But many problems occured and disturb my many days.
my enviroment: Ubuntu 18.04 or centos 7. follow the chipyard tutorial.

First I try to build chipyard in Ubuntu,it work's well.
but when i do 
./marshal -v -d build br-base.json, there have several problems.
i follow the README, Avoid https://github.com/firesim/firesim-software/issues/30 happen.
But when i exectute the above line,the VMware machine crashed the system every time. 
Sometimes tell me it is out of memory.Sometime it doesn't have any output just crashed and restart system.Even I gave the VMware 40GB memory, it's still crashed.

So i turn to Centos 7.
But when i followed the chipyard tutorial to do ./scripts/build-toolchains.sh riscv-tools, it reminds me I need GNU make 4.x。
It seems the tutorials which in https://chipyard.readthedocs.io/en/latest/Chipyard-Basics/Initial-Repo-Setup.html about Centos doesn't work well, but i confirmed that i do those orders.
So i rebuild make ,and then build riscv-tools.
But when i do ./marshal -v -d build br-base.json , Something happened.

DEBUG: make[3]: *** No rule to make target '../lib/libext2fs.so', needed by 'e2fsck'.  Stop.
DEBUG: make[3]: *** Waiting for unfinished jobs....
DEBUG: make[3]: Leaving directory '/home/ljq/chipyard/software/firemarshal/boards/default/distros/br/buildroot/output/build/e2fsprogs-1.45.2/e2fsck'
DEBUG: make[2]: *** [Makefile:434: all-progs-recursive] Error 1
DEBUG: make[2]: Leaving directory '/home/ljq/chipyard/software/firemarshal/boards/default/distros/br/buildroot/output/build/e2fsprogs-1.45.2'
DEBUG: make[1]: *** [Makefile:368: all] Error 2
DEBUG: make[1]: Leaving directory '/home/ljq/chipyard/software/firemarshal/boards/default/distros/br/buildroot/output/build/e2fsprogs-1.45.2'
DEBUG: make: *** [package/pkg-generic.mk:241: /home/ljq/chipyard/software/firemarshal/boards/default/distros/br/buildroot/output/build/e2fsprogs-1.45.2/.stamp_built] Error 2
TaskError - taskid:/home/ljq/chipyard/software/firemarshal/images/br.79c0.img
PythonAction Error
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/doit/action.py", line 437, in execute
    returned_value = self.py_callable(*self.args, **kwargs)
  File "/home/ljq/chipyard/software/firemarshal/boards/prototype/distros/br/br.py", line 190, in buildBaseImage
    wlutil.run(['make'], cwd=br_dir / "buildroot", env=env)
  File "/home/ljq/chipyard/software/firemarshal/wlutil/wlutil.py", line 498, in run
    raise sp.CalledProcessError(p.returncode, prettyCmd)
subprocess.CalledProcessError: Command 'make' returned non-zero exit status 2.

ERROR: Failed to build workload br-base.json

it happened before when i use the AWS instance.i have an issue in firesim when its happen.https://github.com/firesim/firesim/issues/861

So i think the rebuild make cause this problem. But it is a Dead Cycle. IF I do not rebuild make ,I wouldn't build riscv-tools.

I want to ask How to fix it? or How did your guys get the FPGA prototype ？

L-Chocoland, 2021-11-08:
To precisely,My goal is to run linux on VCU118.
I have been generate bitstream in Vivado followed the chipyard tutorial.
And first I generated br-base-no-disk-flat in AWS instance,using firemarshal.
But i got a problem when run linux on VCU118.
When the screen shows:
MIDELEG : 0x0000000000000222
MEDELEG : 0x000000000000b109
PMP0 : 0x0000000080000000-0x000000008001ffff (A)
PMP1 : 0x0000000000000000-0x00000001ffffffff (A,R,W,X)
Then it does not output any things.
follow the issue: #957
i think something wrong in the Firemarshal image.So i think it should be down in my environment,not Firemarshal in AWS instance and ec2fast tools.
so I want to get FPGA prototype in my VMware machine and meet problem i talked above.
hoping for reply

L-Chocoland, 2021-11-08:
same question in  https://github.com/ucb-bar/chipyard/issues/1039

NathanTP, 2021-11-08:
So a few issues are going on.

The first is about trying to run Marshal in a VM. Unfortunately as you've seen, mounting filesystems can be tricky. If you can, make sure you have passwordless sudo enabled for your user which should hopefully avoid the guestmount issue. I have been able to run it in a container before with some careful docker configs, but I've never tried in a VM.

The next issue is general chipyard setup. We have some scripts to help setup centos7 in firesim ([here](https://github.com/firesim/firesim/blob/master/build-setup-nolog.sh)) that you might look at. For ubuntu, I've always followed the instructions on the [chipyard docs](https://chipyard.readthedocs.io/en/latest/Chipyard-Basics/Initial-Repo-Setup.html#requirements) for installing Ubuntu dependencies. Your other issues with marshal build could be related to that. For example, the build-setup.sh script also installs e2fsprogs which I think is what's causing your build error.

Finally, there is a question of whether the marshal images will actually work with the fpga prototype (which seems like your final error). This gets tricky because uarts tend to be touchy about baud rates and things, as well as other differences that make Linux boot difficult. I'll let the folks on the chipyard mailing list help with that since I don't have any experience with FPGA prototypes. Marshal is mostly tested with spike/qemu and the firesim-oriented chipyard builds.

L-Chocoland, 2021-11-13:
@NathanTP Thanks!
I have followed your advise,and have a good result.
so first I use Ubuntu18.04 VMware, and have passwordless sudo for my user, it works well.But the Memory is still required more than 32G,instead I use 48G to make the br-base.json builded successfully. Until now I have not try in Centos,  but i will try in the future.
And then I follow the chipyard Guide in Ubuntu and test the VCU118 and marshal images.it works well and linux can started.
So The main issue I have met is the image which build in marshal.If I want to build in FPGA, the chipyard shoule worked in riscv-tools or esp-tools,not in ec2fast tools.Maybe it can warning other person.
thanks for your reply again! i will close the issue.

closed 2021-11-13 by L-Chocoland

codoor, 2022-04-05:
I don't understand what you do to solve this issue. Can you elaborate on how to solve it? 
I met the same issue recently.
1. For the Firemarshal image, I also see the `No rule to make target '../lib/libext2fs.so', needed by 'e2fsck'. Stop`, but I found the file `libext2fs.so` is in the dir `'../lib/`. I do not konw why it can not found this file.
2. I did not build a Firemarshal image, but my friend gave me a  Firemarshal image which he built. When I open the vcu118 board, the boot process broke, just like your issue
 
[image]

NathanTP, 2022-04-05:
I can't be sure where the e2fsprogs issue is coming from. It's in buildroot rather than marshal directly. For @L-Chocoland, the issue seemed to be down to dependencies on the host (correct me if I'm wrong). You may want to go through the install scripts I listed above and make sure you have all the right packages. Buildroot is also not very good at cleaning up after itself. You may want to try running "make clean" in boards/firechip/distros/br/buildroot. Maybe even go as far as "git clean -fdX" in the buildroot submodule.

As for the existing image not working on vcu118, I'll leave that for @L-Chocoland or the chipyard mailing lists. I'm not too familiar with that process.