firesim/FireMarshal #219: erro: pathspec 'rescv-tools' did not match any file(s) known to git

adnankhalid710, 2021-11-23:
I am using/running firesim from there
 https://github.com/nvdla/firesim-nvdla
Run this command, 
./build-setup.sh fast
facing this error
erro: pathspec 'rescv-tools' did not match any file(s) known to git

can anyone help me on this please?

NathanTP, 2021-11-23:
This issue seems to be related specifically to firesim-nvdla rather than mainline FireSim or FireMarshal. I recommend filing a bug there.

I did notice that the firesim-nvdla repo seems to be 3 years old and doesn't look maintained. A lot has happened to rocketchip and firesim since then. If you're interested in running nvdla, the [chipyard mailing list](https://groups.google.com/g/chipyard) may be able to help (though maybe not, I don't know if anyone has run nvdla recently). Chipyard is a super-project that brings together rocketchip, firesim, and firemarshal (along with other SoC development stuff).

closed 2021-11-23 by NathanTP

adnankhalid710, 2021-11-24:
> This issue seems to be related specifically to firesim-nvdla rather than mainline FireSim or FireMarshal. I recommend filing a bug there.
>
> I did notice that the firesim-nvdla repo seems to be 3 [...]

Hi @NathanTP 
Glad to see your reply. Actually I am new in this filed (with software background), thanks for your guidance. let me dive in it.
Regards and best wishes @NathanTP .
Adnan