firesim/FireMarshal #229: crashs while building br-base.json

Flipfle, 2022-02-15:
I want to use the firemarshal to work with NVDLA in Chipyard.

At first I wanted to test firemarshal and executed the command 
`./marshal -v -d build br-base.json`

The whole machine crashes while running:
`Running: "make ARCH=riscv CROSS_COMPILE=riscv64-unknown-linux-gnu- vmlinux Image -j" in /home/parallels/Documents/chipyard/software/firemarshal/boards/firechip/base-workloads/br-base/../../linux`

The last lines of the log are:
```
2022-02-15 10:30:31,142 [run         ] [DEBUG]  CC      drivers/spi/spi-sifive.o
2022-02-15 10:30:31,248 [run         ] [DEBUG]  CC      fs/ext4/mballoc.o
2022-02-15 10:30:31,248 [run         ] [DEBUG]  CC      drivers/base/power/common.o
2022-02-15 10:30:32,921 [run         ] [DEBUG]  CC      drivers/gpu/drm/radeon/radeon_drv.o
2022-02-15 10:30:33,115 [run         ] [DEBUG]  CC      drivers/scsi/scsi_ioctl.o
2022-02-15 10:30:33,351 [run         ] [DEBUG]  CC      drivers/pci/remove.o
2022-02-15 10:30:33,412 [run         ] [DEBUG]  CC      drivers/gpu/drm/radeon/radeon_asic.o
2022-02-15 10:30:33,477 [run         ] [DEBUG]  CC      drivers/pci/pci.o
2022-02-15 10:30:33,713 [run         ] [DEBUG]  CC      drivers/gpu/drm/radeon/radeon_kms.o
2022-02-15 10:30:33,839 [run         ] [DEBUG]  CC      drivers/scsi/scsicam.o
2022-02-15 10:30:33,849 [run         ] [DEBUG]  AR      drivers/gpu/drm/omapdrm/built-in.a
2022-02-15 10:30:33,978 [run         ] [DEBUG]  CC      drivers/scsi/scsi_error.o
2022-02-15 10:30:34,033 [run         ] [DEBUG]  CC      net/ethtool/ioctl.o
2022-02-15 10:30:34,304 [run         ] [DEBUG]  CC      drivers/pci/pci-driver.o
2022-02-15 10:30:35,713 [run         ] [DEBUG]  CC      net/core/sock.o
2022-02-15 10:30:50,326 [run         ] [DEBUG]  CC      net/core/request_sock.o
```

I installed Chipyard like in the documentation described. In Firemarshal, I installed the python requirements and I initialised the submodules. In `boards/default/linux` I checkout the branch to `nvdla-linux-v57`.

I work on Ubuntu in a virtual machine. 
'Linux parallels-Parallels-Virtual-Platform 5.13.0-27-generic #29~20.04.1-Ubuntu SMP Fri Jan 14 00:32:30 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux'

I have enabled sudo to passwordless for my user (#215).

Does Firemarshal simply not work on a VM?
I am new to this subject so it could be that I missed something important, I just don't know what. :)

NathanTP, 2022-02-15:
This is most likely caused by an old bug with marshal where we messed up the jlevel. This was fixed in #207 and is included in the latest 1.12.1 release (which I just pushed yesterday).

Flipfle, 2022-02-17:
thank you, that resolved my issue!

closed 2022-02-17 by Flipfle