firesim/FireMarshal #232: Bump buildroot submodule version to pull in upstream fixes

allisonrandal, 2022-02-21:
In the process of testing other dependency updates for Chipyard (https://github.com/ucb-bar/chipyard/pull/1120), I discovered some build failures in FireMarshal on Ubuntu 21.10. Currently, the FireMarshal README mentions Ubuntu 18.04, and the FireSim docs mention Centos 7.7, so these dependency version incompatibilities aren't immediately relevant, but this issue captures the relevant details so you have them when you need them.

The first failure has already been reported in #30, and that workaround has to be applied before the subsequent failures will appear.

The second failure is that buildroot fails to build images with errors "You must install 'python' on your build machine". This has already been fixed upstream in buildroot: The buildroot changelog for 2022.02-rc1 notes: "Python 2.7 and python 2.x specific packages removed as python 2.x is EOL since April 2020." (The specific relevant change is https://github.com/buildroot/buildroot/commit/9c0c7846cdb63f3ed62563c4f532ad02ad0a1c98). The short-term workaround for this error is to create a symlink named 'python' pointing to 'python2' (Ubuntu stopped shipping a binary named 'python' years ago, and only ships binaries named 'python3' and 'python2').

The third failure in building images is a C compile error in fakeroot, which is pulled in as a dependency by buildroot. The problem has already been fixed upstream, first by patching buildroot (https://github.com/buildroot/buildroot/commit/f45925a951318e9e53bead80b363e004301adc6f) and later by updating the version of fakeroot that buildroot pulls in as a dependency (https://github.com/buildroot/buildroot/commit/91ebf2b58b119a43134a95994c3cd63297f599fa). 

Bumping the git submodule for buildroot to a new version would resolve the second and third failures, but it needs testing on older versions of Centos to make sure the dependency update doesn't break the current AMIs recommended for use in ec2.

allisonrandal, 2022-02-21:
Attaching log files from scripts/fullTest.py, showing examples of the Python and C errors.

[bbl-test-2022-02-18--17-31-32-LWWMFJYCJUQ2C9AR.log](https://github.com/firesim/FireMarshal/files/8110542/bbl-test-2022-02-18--17-31-32-LWWMFJYCJUQ2C9AR.log)
[bbl-test-2022-02-18--18-02-53-MK6SFFCNYAHL3HVV.log](https://github.com/firesim/FireMarshal/files/8110543/bbl-test-2022-02-18--18-02-53-MK6SFFCNYAHL3HVV.log)

NathanTP, 2022-02-21:
#30 is unfortunately fairly intractable unless and until we find a more portable solution to manipulating guest filesystems (which I would love, but don't really have time for).

I'm fine with bumping buildroot in principle. In theory that shouldn't be a huge deal (I've done some work to ease that process) but we don't do it often so sneaky stuff might crop up. I know at least the sha3 tutorial workload depends on some buildroot-specific stuff. Maybe a few other things as well. Again, FireMarshal is architected to limit that impact, but you never know. I'd be happy to accept a PR for that. It might be awhile before I can get to it myself.

PR #237 (2022-09-20): bump buildroot to 2022.02
I was able to build br-base.json just fine without any other changes to marshal.  I did run into the 'open files' ulimit.  I don't ever remember running into that and I don't think that firesim's machine-launch-script.sh increases the ulimit at all.  Maybe we should?

I'm running firesim/deploy/regression.sh on the default AGFI's.  @NathanTP if there is anything else I should run to stress test things after bumping buildroot, I'd be happy to.

fixes #232 
fixes #236
closed 2022-09-20 by abejgonzalez