firesim/FireMarshal #290: ERROR: Failed to build workload br-base.json

zzulb, 2023-10-11:
I encountered an error while using chipboard and I don't know how to solve it!

DEBUG: Running: "e2fsck -f -p /chipyard/software/firemarshal/images/firechip/br-base/br-base.img" in /chipyard/software/firemarshal
DEBUG: /chipyard/software/firemarshal/images/firechip/br-base/br-base.img 已挂载。
DEBUG: e2fsck: 无法继续，已中止。
DEBUG: 
DEBUG: 
TaskError - taskid:/chipyard/software/firemarshal/images/firechip/br-base/br-base.img
PythonAction Error
Traceback (most recent call last):
  File "/chipyard/.conda-env/lib/python3.10/site-packages/doit/action.py", line 461, in execute
    returned_value = self.py_callable(*self.args, **kwargs)
  File "/chipyard/software/firemarshal/wlutil/build.py", line 640, in makeImage
    wlutil.resizeFS(config['img'], 0)
  File "/chipyard/software/firemarshal/wlutil/wlutil.py", line 625, in resizeFS
    raise sp.CalledProcessError(ret, " ".join(chkfsCmd))
subprocess.CalledProcessError: Command 'e2fsck -f -p /chipyard/software/firemarshal/images/firechip/br-base/br-base.img' returned non-zero exit status 8.

ERROR: Failed to build workload br-base.json
Log available at: /chipyard/software/firemarshal/logs/br-base-build-2023-10-11--13-39-44-MJ7EXEW4UMYON35S.log
ERROR: FAILURE: 1 builds failed

JimmyWong1998, 2023-10-22:
Hi zzulb, I encounter the same error. Have you solved this issue?

Jimmy

zzulb, 2023-10-22:
Yes, you can unmount the br-base.img image. And then re execute the command.