firesim/FireMarshal #309: ./build-setup.sh stops at step 9 (re-mounting the disk-mount folder)

wadidf, 2024-08-14:
https://github.com/firesim/FireMarshal/blame/74ac78a0b9ae106dff9c5bdfc3a89847ae11d357/wlutil/wlutil.py#L598C12-L598C12

`./build-setup.sh riscv-tools` stops at step 9
the cause is the assertion in line 598 (wlutil.py) regarding the mount-point of **disk-mount** folder.
Well the folder is already mounted before, and the assertion will stop the setup script.
As a workaround I commented the line and re-run the build script after sourcing the environment (to have $RISCV defined) and skipping the steps : 1 to 8. 
But it would be efficient to replace the assertion by a test and warning.

cavendish98, 2024-08-20:
I have this problem too, how did you solve it?

abejgonzalez, 2024-08-29:
I think this should be fixed now. There was an issue w/ Ubuntu 22 versions of `mountpoint` returning a different error code than Ubuntu 20 versions.

cavendish98, 2024-09-15:
> I think this should be fixed now. There was an issue w/ Ubuntu 22 versions of `mountpoint` returning a different error code than Ubuntu 20 versions.

no,it didn't work

xiuhu17, 2024-09-22:
Same issue.....

wadidf, 2024-09-22:
It is fixed in Firemarshal (in the code :wlutil.py lines 598-599)
```
# mountpoint on Ubuntu 20.* returns 1 (on 22.* it returns 32) for an empty folder
    assert ret == 1 or ret == 32, f"{mntPath} already mounted. Somethings wrong"
```

But Chipyard is not yet fetching the updated version.
@xiuhu17 and @cavendish98, as a workaround, you can apply yourself the modification (the assert line) 
then in the (same) terminal run the commands :
```
source  env.sh
./build-setup.sh riscv-tools -s 1 -s 2 -s 3 -s 4 -s 5 -s 6 -s 7 -s 8
```

xiuhu17, 2024-09-22:
> It is fixed in Firemarshal (in the code :wlutil.py lines 598-599)
> 
> ```
> 598     # mountpoint on Ubuntu 20.* returns 1 (on 22.* it returns 32) for an empty folder
> 599        assert ret == 1 or ret == 32, f"{mntPath} already mounted. Somethings wrong"
> ```
> 
> But Chipyard is not yet fetching the updated version. @xiuhu17 and @cavendish98, as a workaround, you can apply yourself the modification (the assert line) then in the (same) terminal run the commands :
> 
> ```
> source  env.sh
> ./build-setup.sh riscv-tools -s 1 -s 2 -s 3 -s 4 -s 5 -s 6 -s 7 -s 8
> ```

Thanks for letting me know :)

sowrabh-adiga, 2024-12-30:
> It is fixed in Firemarshal (in the code :wlutil.py lines 598-599)
>
> ```
> 598     # mountpoint on Ubuntu 20.* returns 1 (on 22.* it returns 32) for an empty folder
> 599        assert ret == 1 or ret == [...]

 Thanks @wadidf ! was stuck at the same step, and your workaround worked perfectly for me.