firesim/FireMarshal #312: Building br-base.json for prototyping fails (unable to download cached image)

ktlpdlbr, 2024-10-22:
Hi I am following the instructions given in [chipyard documentation](https://chipyard.readthedocs.io/en/latest/Prototyping/VCU118.html#building-linux-with-firemarshal) to build linux image for my FPGA prototype however building it I have following logs: 

` [buildBaseIma] [INFO ]  Attempting to download cached image: https://raw.githubusercontent.com/firesim/firemarshal-public-br-images/main/images/prototype/br.b8b4/br.b8b4.img.zip
 [buildBaseIma] [DEBUG]  urlretrieve exception: HTTP Error 404: Not Found
 [run         ] [DEBUG]  Running: "make defconfig" in /chipyard/software/FireMarshal/boards/prototype/distros/br/buildroot
 [run         ] [DEBUG]  mkdir -p /chipyard/software/FireMarshal/boards/default/distros/br/buildroot/output/build/buildroot-config/lxdialog`

And in the end, it fails with the following logs:

`DEBUG: make[1]: *** [package/pkg-generic.mk:273: /chipyard/software/FireMarshal/boards/default/distros/br/buildroot/output/build/toolchain-external-custom/.stamp_configured] Error 1
DEBUG: make: *** [Makefile:82: _all] Error 2
TaskError - taskid:/chipyard/software/FireMarshal/images/prototype/br.b8b4/br.b8b4.img
PythonAction Error
Traceback (most recent call last):
  File "/chipyard/.conda-env/lib/python3.10/site-packages/doit/action.py", line 461, in execute
    returned_value = self.py_callable(*self.args, **kwargs)
  File "/chipyard/software/FireMarshal/boards/prototype/distros/br/br.py", line 271, in buildBaseImage
    wlutil.run(['make'], cwd=br_dir / "buildroot", env=env)
  File "/chipyard/software/FireMarshal/wlutil/wlutil.py", line 527, in run
    raise sp.CalledProcessError(p.returncode, prettyCmd)
subprocess.CalledProcessError: Command 'make' returned non-zero exit status 2.`

Jerryy959, 2024-10-26:
I have the same issue

Jerryy959, 2024-10-27:
> Hi I am following the instructions given in [chipyard documentation](https://chipyard.readthedocs.io/en/latest/Prototyping/VCU118.html#building-linux-with-firemarshal) to build linux image for my [...]

[image]

I guess you have already running this command, but the correspond repo: https://github.com/firesim/firemarshal-public-br-images/tree/main seems that there is no prototype dir.

Jerryy959, 2024-10-27:
and if i delete the file <marshal-config.yaml>, everything works fine.

panjin12345, 2024-11-10:
I have the same problem, here is my log report:
[br-base-build-2024-11-10--11-27-20-SC1LFS8KCDQ28IRT.log](https://github.com/user-attachments/files/17691610/br-base-build-2024-11-10--11-27-20-SC1LFS8KCDQ28IRT.log)

Jerryy959, 2024-11-11:
@abejgonzalez @NathanTP

ipapal, 2025-03-21:
Is there any update?

rfgIII, 2026-03-02:
Is there an update on this issue? I cleared the cache in the buildroot and ran the fix-open-files.sh to check the limit on my system. Still running into the same issue.

CirbuildProject, 2026-10-06:
Installing through Chipyard 
Similar Error on br-base.json build, but root cause could be different
Log shows that the default assigned mountpoint directory is NOT a mountpoint, and attempts to fetch the guestmount
(I do not understand the problem)
2026-10-06 17:54:28,269 [run         ] [DEBUG]  Running: "make -q LINUXSRC=/home/takeshi/chipyard/software/firemarshal/boards/firechip/base-workloads/br-base/../../linux" in /home/takeshi/chipyard/software/firemarshal/boards/firechip/base-workloads/br-base/../../drivers/icenet-driver
2026-10-06 17:54:28,273 [run         ] [DEBUG]  Running: "make -q LINUXSRC=/home/takeshi/chipyard/software/firemarshal/boards/firechip/base-workloads/br-base/../../linux" in /home/takeshi/chipyard/software/firemarshal/boards/firechip/base-workloads/br-base/../../drivers/iceblk-driver
2026-10-06 17:54:28,275 [handleHostIn] [DEBUG]  Applying host-init: /home/takeshi/chipyard/software/firemarshal/boards/firechip/base-workloads/br-base/host-init.sh
2026-10-06 17:54:28,275 [run         ] [DEBUG]  Running: "/home/takeshi/chipyard/software/firemarshal/boards/firechip/base-workloads/br-base/host-init.sh" in /home/takeshi/chipyard/software/firemarshal/boards/firechip/base-workloads/br-base
2026-10-06 17:54:28,277 [run         ] [DEBUG]  make: Entering directory '/home/takeshi/chipyard/software/firemarshal/boards/firechip/base-workloads/br-base/trigger'
2026-10-06 17:54:28,277 [run         ] [DEBUG]  make: Nothing to be done for 'all'.
2026-10-06 17:54:28,277 [run         ] [DEBUG]  make: Leaving directory '/home/takeshi/chipyard/software/firemarshal/boards/firechip/base-workloads/br-base/trigger'
2026-10-06 17:54:28,718 [makeImage   ] [DEBUG]  Applying overlay: /home/takeshi/chipyard/software/firemarshal/boards/firechip/base-workloads/br-base/overlay
2026-10-06 17:54:28,718 [run         ] [DEBUG]  Running: "mountpoint /home/takeshi/chipyard/software/firemarshal/disk-mount" in /home/takeshi/chipyard/software/firemarshal
2026-10-06 17:54:28,720 [run         ] [DEBUG]  /home/takeshi/chipyard/software/firemarshal/disk-mount is not a mountpoint
2026-10-06 17:54:28,723 [run         ] [DEBUG]  Running: "guestmount --pid-file ./guestmount.pid -o uid=1000 -o gid=1000 -a /home/takeshi/chipyard/software/firemarshal/images/firechip/br-base/br-base.img -m /dev/sda /home/takeshi/chipyard/software/firemarshal/disk-mount" in /home/takeshi/chipyard/software/firemarshal
2026-10-06 17:54:28,757 [main        ] [ERROR]  Failed to build workload br-base.json
2026-10-06 17:54:28,757 [main        ] [INFO ]  Log available at: /home/takeshi/chipyard/software/firemarshal/logs/br-base-build-2026-10-06--09-54-27-V3J5KYBUAGN8R6SZ.log
2026-10-06 17:54:28,757 [main        ] [ERROR]  FAILURE: 1 builds failed